PX4/PX4-Autopilot #24850: [Bug] Switching to Precision Land

TeddyZaremba, 2025-05-14:
### Describe the bug

The drone went straight up when I switched to precision land.

### To Reproduce

Take vehicle off in position mode
Switch to precision land without any April Tags beneath it

### Expected behavior

Should land or do nothing

### Screenshot / Media

https://www.loom.com/share/ea83c73f2f514ffd9b7796aff0b2edad?sid=46096409-79c4-4ad6-916e-57b2d1fd02e0

### Flight Log

https://review.px4.io/plot_app?log=162e8f7d-2019-47b0-aa60-7c6c221f575a

### Software Version

QGC v4.4.4
PX4 v1.16.0 (RC)

### Flight controller

VOXL2

### Vehicle type

None

### How are the different components wired up (including port information)

Starling 2 Max

Position: GPS
ESC: VOXL ESC Mini 4-in-1
Battery: 2s 3000mah

dakejahl, 2025-05-14:
It's PLD_SRCH_ALT

TeddyZaremba, 2025-05-14:
> It's PLD_SRCH_ALT

Ah makes sense

afwilkin, 2025-05-28:
@TeddyZaremba I think we can close this correct?

closed 2025-05-28 by TeddyZaremba