PX4/PX4-Autopilot #24857: [Docs] [Bug] Radio Control docs don't reflect new architecture

hamishwillee, 2025-05-15:
### Describe the bug

RC setup seems to be a bit confusing currently. 

> [!NOTE] Bunch of questions/confirmations here. If I can get confirmation, I can update the docs. @dagar > @dakejahl If you could answer the questions would be great. I think most of it should be "yes, that is correct"

I think the TLDR is that if you want to use RC you should:
- Check the new RC modules are in firmware. Easily enough done by looking at their params in QGC, such as `RC_SBUS_PRT_CFG`
- You'd enable the param you want on the port labeled  "Radio Control" (presumably) and check the others are disabled, including the Rc input param.
- That should be enough to enable normal RC if you connect a receiver of the appropriate type.

Is that about right? It may be that if you also want telemetry you also have to enable the seaparate params for CRSF and GHST (presumably on the same port).

MOre detailed context below - additional things like "what new protocol module supports ST24?"

----

Is all this right. Please highlight bits that are wrong?

1. Historically we had a single module rc_input that was present in all board firmware, and which was started on boot to connect to whatever port is the "RC Input" or SBUS or whatever.
   You would attached RC to the physical input of choice and it would auto-detect your protocol.
   a. The board has PPM next to the PWM outputs, and often has separate DSM/SBUS/RSSI port. Does this mean we ran multiple instances of the module pointing to each connector?
   b. Is the physical RC ports compatible with all the protocols? (PPM, SBUS, DSM. SUMD, ST24, TBS Crossfire (CRSF))?
   c. All of these are strict "RC input" right? I.e. they are not telemetry too?

2. We also had  Frsky Telemetry (Listed as communications driver) and CRSF listed as an RC protocol - but still doing telemetry only.
   a. My understanding is CRSF module works with  TBS Crossfire radio system or ExpressLRS radio systems, and sends both telemetry and RC information.
      This is somehow different from "normal TBS crossfire CRSF" RC right? In what way?
   b. Previously we had to remove RC-in from firmware to add CRSF - this doesn't appear necessary.

3. We decided to split up the monolithic rc-input module into modules for separate protocols.
   This removes the complication of auto-detection, and makes it possible to produce smaller firmware (in theory).
  
    - The new modules are added to most boards by default using `CONFIG_COMMON_RC=y`.
    - The old rc_input driver is added to some boards using `CONFIG_DRIVERS_RC_INPUT=y`
    - Some have both keys.

4. The intent is that eventually when all boards have the new modules configured as desired the old rc_input will be removed from firmware everywhere, and eventually from the source tree.

5. Currently we have these modules:
   - [crsf-rc](https://docs.px4.io/main/en/modules/modules_driver.html#crsf-rc) - parses the CRSF RC uplink protocol and generates CRSF downlink telemetry data
   - [dsm-rc](https://docs.px4.io/main/en/modules/modules_driver.html#dsm-rc) - This module does Spektrum DSM RC input parsing.
   - [ghst-rc](https://docs.px4.io/main/en/modules/modules_driver.html#ghst-rc) - This module does Ghost (GHST) RC input parsing.
   - [sbus-rc](https://docs.px4.io/main/en/modules/modules_driver.html#sbus-rc) - This module does SBUS RC input parsing.
   
6. Which of those above are used for which RC protocols? 
    I mean I can guess from names which do DSM, TBS Crossfire (CRSF), GHST. But which do PPM, SBUS, . SUMD, ST24?

7. These protocols are assigned to a port using the following parameters. 
     - [RC_SBUS_PRT_CFG](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#RC_SBUS_PRT_CFG) - port for SBUS (default is 300 ("Radio Controller"), but may be overridden)
     - [RC_CRSF_PRT_CFG](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#RC_CRSF_PRT_CFG) - 0 (disabled)
      - [RC_DSM_PRT_CFG](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#RC_DSM_PRT_CFG) - 0 (disabled)
     - [RC_GHST_PRT_CFG](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#RC_GHST_PRT_CFG) - 0 (disabled)

   1. Do we need to do anything other than assign these to a port to get them to work? (other than disable anything assigned)?

8. What are these params for then? 
        - [RC_CRSF_TEL_EN](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#RC_CRSF_TEL_EN)
        - [RC_GHST_TEL_EN](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#RC_GHST_TEL_EN)
		
      I assume that Telemetry and RC enabling all move on the same port, but for some reason you might choose to sometimes just send RC. Why?

9. And what is this one for then?
   - TEL_BST_EN (INT32) Blacksheep telemetry Enable. If true, the FMU will try to connect to Blacksheep telemetry on start up (looks like duplicate of RC_CRSF_TEL_EN

10. I assume these are separate modules you can enable for different kinds of telemetry that is separate from the RC?
   - TEL_FRSKY_CONFIG (INT32) Serial Configuration for FrSky Telemetry. Configure on which serial port to run FrSky Telemetry.
   - TEL_HOTT_CONFIG (INT32) Serial Configuration for HoTT Telemetry.

hamishwillee, 2025-05-18:
- [ ] How is the `300: Radio Communication` port mapped?
  - When I try using the feature on my v6x board, I don't get offered all ports, and in particular the  `300: Radio Communication` mapped to the port you should assign. 
  - This also isn't something we map in the docs - https://docs.px4.io/main/en/flight_controller/cuav_pixhawk_v6x.html#serial-port-mapping . How can we work out what port it is to map from the board config?
  
@dagar ?

julianoes, 2025-06-05:
> 1.a. The board has PPM next to the PWM outputs, and often has separate DSM/SBUS/RSSI port. Does this mean we ran multiple instances of the module pointing to each connector?

No, either one module using more than one microcontroller pin (e.g. [ARK 6X configuration (without IO)](https://github.com/PX4/PX4-Autopilot/blob/ff2b82dc51cf0d2282a10421db9338572725feaf/boards/ark/fmu-v6x/src/board_config.h#L322-L327)). 

Or recently combined in one microcontroller pin (e.g. [Kakute H7 Wing](https://github.com/PX4/PX4-Autopilot/blob/ff2b82dc51cf0d2282a10421db9338572725feaf/boards/holybro/kakuteh7-wing/src/board_config.h#L174-L175)).

>  1.b. Is the physical RC ports compatible with all the protocols? (PPM, SBUS, DSM. SUMD, ST24, TBS Crossfire (CRSF))?

For FMU: all of them (see [rc_input](https://github.com/PX4/PX4-Autopilot/blob/ff2b82dc51cf0d2282a10421db9338572725feaf/src/drivers/rc_input/RCInput.hpp#L94-L101)).
For IO: PPM, SBUS, DSM, ST24 (see [px4iofirmware](https://github.com/PX4/PX4-Autopilot/blob/ff2b82dc51cf0d2282a10421db9338572725feaf/src/modules/px4iofirmware/controls.cpp#L46-L50))

> 1.c. All of these are strict "RC input" right? I.e. they are not telemetry too?

For FMU they might have telemetry, for IO it's just RC input as far as I know.

> 2. CSRF, FrSky Telemetry?

You tell me. I don't know the details.

> 3. This removes the complication of auto-detection, and makes it possible to produce smaller firmware (in theory).

That's correct and also the goal is to just get rid of all auto-detection magic and rather have a declarative setup instead. The focus is on one specific locked down config, rather than a do-it-all approach. It removes a lot of failure cases which auto-detection can mask. Without auto-detection you can specifically complain if a GPS or RC is missing because you know this (and exactly this) should be available rather than "meh I tried everything but couldn't detect anything".

The inconsistency in terms of what you see in default.px4board is just because of the FMU vs IO split, or just boards that aren't up to date.

> 4. The intent is that eventually when all boards have the new modules configured as desired the old rc_input will be removed from firmware everywhere, and eventually from the source tree.

For @dagar to confirm but it sounds about right.

> 6. But which do PPM, SBUS, . SUMD, ST24?

Your guess is as good as mine. I assume SBUS, SUMD, ST24 don't exist yet. And PPM does neither but that's because it needs a timer pin and not a serial pin, so it's not necessarily available on any serial port.

> 7.i. Do we need to do anything other than assign these to a port to get them to work?

I don't think so but you don't know until you try.

> 8. but for some reason you might choose to sometimes just send RC. Why?

I assume it goes back to the "we specify everything declaratively" paradigm.

> 9. Blacksheep telemetry Enable.

[Looks like it starts bst](https://github.com/search?q=repo%3APX4%2FPX4-Autopilot%20TEL_BST_EN&type=code). [Some code from 2015](https://github.com/PX4/PX4-Autopilot/blob/ff2b82dc51cf0d2282a10421db9338572725feaf/src/drivers/telemetry/bst/bst.cpp), not sure.

> 11. Whats the difference between RC_PORT_CONFIG and RC_INPUT_PROTO

This confused me as well recently!

- [RC_PORT_CONFIG](https://docs.px4.io/v1.15/en/advanced_config/parameter_reference.html#RC_PORT_CONFIG) chooses the serial port for rc_input.
- [RC_INPUT_PROTO](docs.px4.io/v1.15/en/advanced_config/parameter_reference.html#RC_INPUT_PROTO) selects the protocol (e.g. SBUS).

Hope that helps. Just in terms of docs, is this more user documentation or developer documentation? The questions seem geared towards developer documentation but the parameters itself would be user facing. And I'm not sure it's worth documenting the developer docs too closely as it's obviously all in flux. I think it's worthwhile to highlevel document what the goal is and how it is changing but not necessarily each and every protocol and feature. At least for me, I would just look at the parameters and code and try it out directly as I would never expect docs documenting implementation to be in sync, no offense...

hamishwillee, 2025-06-11:
> > My theory is ports are only assignable if they are on the FMU. So the absence of 300: Radio Controller indicates that the RC port on this particular board is done though the IO.
> 
> I believe `300: Radio Controller` is shown when [this line is in the default.px4board](https://github.com/PX4/PX4-Autopilot/blob/main/boards/ark/fmu-v6x/default.px4board#L10). And - I think - if it is configured correctly that's the case.

@julianoes I think it might be a little more complicated. The CUAV Pixhawk v6x appears to use the default.px4board - or at least it downloads the same name firmware. However I am pretty sure this has an IO board and I am not offered Radio Controller as an option for selection when using main.

So I think it is "must have this assignment" AND no IO board. 

Would it be that `"/dev/ttyS5"` doesn't map to anything on an FMU v6x board with an IO board so this isn't displayed?

Does Radio Controller get displayed for any board you have? It doesn't work for anything I have, which makes me worry that perhaps my QGC set or something else is wrong.

hamishwillee, 2025-06-11:
I think I might be able to write this now. Though it is going to be messy and possibly inaccurate in the fine detail.

hamishwillee, 2025-06-11:
@julianoes That's worse than I thought.

> * IO available, RC connected to FMU (CUAV 6X?)

CUAV 6X does not show `Radio Communication` as an option. My assumption therefore is that this is the case of `> * IO available, RC connected to IO`

As in my last comment I suspect that the theory is that FMUv6x might be used in IO and mini versions, so you would get that option in the IO-less version.

> * IO available, RC connected to both (not sure which ones have it that way, maybe Holybro/PX4 6X?)

Specifically, you don't have any control over what is running on the IO to turn off RC say, so wouldn't this always result in clashes?
How could this work?

> That's why it's messy and I'm almost thinking we need to go through all boards and update their docs and also create some overview table.

Yes. 

Again though, knowing if you can check  `Radio Communication` as an option in QGC to know if the RC is FMU connection would make this a lot easier. 

Do we know **any** board that allows you to see this option? I have tried Cuav 6x, 7 nano, Pixhawk 4 mini. The later doesn't even have an IO board.

hamishwillee, 2025-06-11:
Right now  the instructions would have to be 
- If you want to use one of the older protocols plug your receiver into the RC port. If QGC shows that RC is working then your RC is configured.
- If that doesn't work you **can't use the RC port on your Pixhawk**. Pick some other unused port and do X, Y, Z.

julianoes, 2025-06-12:
I'd have to go through all the boards I have and do some research.

julianoes, 2026-08-31:
Leaving this open, I think it's still a mess.

hamishwillee, 2026-08-31:
It is, thanks. The problem is that there is _no good solution_. I know how to improve it a bit and will do so when I have time.

dakejahl, 2026-08-31:
Solution
1. Kill `rc_input`
2. Switch from per-protocol mapping to per-serial mapping https://github.com/PX4/PX4-Autopilot/issues/25004
Compile time metadata generation from KConfig. Every driver than can be started on a serial port would need to declare itself as such so that we can build the list. The thought is that this would be analagous to ardupilot `SERIALn_PROTOCOL` but instead we'd replace `SERIALn_PROTOCL` with for example `TELEM1_PROTOCOL`. S3 would have everything. GCS would show everything until metadata from FC is downloaded.
3. Write the param translation (or accept breaking change for PX4 2.0). 
4. Point Claude at this comment and tell it to do it.
5. Hook up a FC on your bench with an stlink swd/uart to the FC. Connect a mavlink stream, RC, and GPS to whatever ports you want. 
6. Tell Claude to test the PR it just wrote. Give it permission to use the stlink, power control on USB if your USB port allows it, or a UART controlled PSU like the Rigol DP712. 

There's the RC over PX4IO MCU we need to consider for some carriers. In that case usually it's USART6 (RC) that is connected to the PX4IO. The `input_rc` topic is published from PX4IO and protocol support is tied to the PX4IO firmware. 

^ I'm probably wrong or missing information to some degree. Claude should check my facts and push back if what I'm proposing won't work. 

By making this switch to per-serial mapping it provides:
- foot-gun free way to configure serial ports
- the same parameter workflow for users who use both Ardupilot/PX4
- allows us to build a **Serial Port Configuration** page in QGC reusing all of the same Qml code since the mapping convention is the same. 

:)

julianoes, 2026-09-01:
I was going to say. The solution is to clean it up, then to fix the docs

dakejahl, 2026-09-01:
Grok is working on it https://github.com/PX4/PX4-Autopilot/pull/28481

hamishwillee, 2026-09-02:
@dakejahl 
What I think you're proposing is that we get rid of RC_INPUT. How does that help people who are using a board with IO? 

The root of the RC problem is that boards have a labelled RC port that behaves differently depending on whether it is connected to the IO or the FMU, and in the FMU case, whether RC_INPUT or COMMON_RC is installed.
Unless we can present the same behaviour across any RC port, we're necessarily "polishing a turd".