PX4/PX4-Autopilot #24864: [Bug] GPS Accuracy and Fix Type are not processed for Holybro H-RTK UM982

sinancimen, 2025-05-16:
### Describe the bug

GPS accuracy fields eph, epv, h_acc, v_acc and fix_type fields are not processed correctly by PX4. Fix Type is always 3 and others are always zero when I check them through either Mavlink Inspector (RAW_GPS_DATA) or Mavlink Console for UORB Message (sensor_gps). Position and velocities are reported correctly.

I configured the receiver to publish all NMEA messages from UPrecise software and I can see them through data stream of UPrecise. Accuracies and fix types seem to be correct through UPrecise; however, they are not processed by PX4.

### To Reproduce

Anytime when there's a GPS fix.

### Expected behavior

Fields should have been filled.

### Software Version

v1.15.4

### Flight controller

Pixhawk V6X

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

Receiver is connected to GPS1 port of Pixhawk.

julianoes, 2025-05-18:
Any chance you can share a logfile?

sinancimen, 2025-05-19:
I realized that I changed NMEA settings of the receiver for the wrong COM port. Setting it for COM3 fixed the problem. Sorry for the wrong issue.

closed 2025-05-19 by sinancimen