PX4/PX4-Autopilot #24882: [Bug] commander takeoff fails on gz_x500 despite all preflight checks disabled (SITL + Ignition + MicroDDS)

DrChanChan, 2025-05-20:
### Describe the bug

Hello, I’m currently working on a PX4-based simulation project to control multiple UAVs in Ignition Gazebo without any physical hardware.
According to the PX4 guide ([link](https://docs.px4.io/v1.15/en/sim_gazebo_gz/multi_vehicle_simulation.html)), I'm testing a single UAV instance (`gz_x500`) before scaling to a multi-UAV setup. I’m using ROS 2 (Humble) and Micro XRCE-DDS Agent, which is running correctly.
I launch the PX4 instance using:

**PX4_SYS_AUTOSTART=4001 PX4_GZ_MODEL_POSE="0,5" PX4_SIM_MODEL=gz_x500 UXRCE_DDS_KEY=2 MAV_SYS_ID=2 ./build/px4_sitl_default/bin/px4 -i 1**

However, both `commander arm` and `commander takeoff` fail with the following messages:

**pxh> commander arm
WARN [commander] Arming denied: Resolve system health failures first
pxh> commander takeoff
WARN [commander] Arming denied: Resolve system health failures first**

I've tried modifying several params like COM_ARM_WO_GPS 1, NAV_RCL_ACT 0, etc. but it doesn't work. 
There is no solution to this in the guide, so I'm asking here.
I'm using ROS2 humble, PX4_v1.15.
Is there any workaround I can do?
Here's my terminal state in case it helps: left is MicroDDS Terminal, right is gz_x500 manually turned on.

[image]

### Vehicle type

None

dakejahl, 2025-05-20:
Please try the `main` branch

ckjunaid, 2025-06-21:
connect qgc

Himanshu069, 2025-07-05:
I'm also facing the same issue. 
Isn't there any workaround for this?

closed 2026-01-12 by github-actions[bot]