PX4/PX4-Autopilot #24883: [Bug] Issue when running setup "ubuntu.sh" script in ubuntu 22.04

MoonlightShinesparkle, 2025-05-20:
### Describe the bug

When running the setup script for ubuntu i run into a python error which stops the dependency installation requiered for me to setup px4, here is the console output (environment auto enables ROS commands):

> Successfully enabled ROS commands

> robotica@secure-server:~$ '/home/robotica/PX4-Autopilot/Tools/setup/ubuntu.sh' 
> Ubuntu 22.04
> 
> Installing PX4 general dependencies
> [sudo] password for robotica: 
> Hit:1 https://packages.microsoft.com/repos/code stable InRelease
> Hit:2 http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy InRelease
> Hit:3 http://mx.archive.ubuntu.com/ubuntu jammy InRelease
> Hit:4 http://mx.archive.ubuntu.com/ubuntu jammy-updates InRelease
> Hit:5 http://packages.ros.org/ros2/ubuntu jammy InRelease
> Hit:6 http://mx.archive.ubuntu.com/ubuntu jammy-backports InRelease
> Hit:7 http://security.ubuntu.com/ubuntu jammy-security InRelease
> Reading package lists...
> Reading package lists...
> Building dependency tree...
> Reading state information...
> build-essential is already the newest version (12.9ubuntu3).
> g++ is already the newest version (4:11.2.0-1ubuntu1).
> gcc is already the newest version (4:11.2.0-1ubuntu1).
> make is already the newest version (4.3-4.1build1).
> zip is already the newest version (3.0-12build2).
> astyle is already the newest version (3.1-2build1).
> cppcheck is already the newest version (2.7-1).
> lcov is already the newest version (1.15-1).
> libfuse2 is already the newest version (2.9.9-5ubuntu3).
> ninja-build is already the newest version (1.10.1-1).
> shellcheck is already the newest version (0.8.0-2).
> cmake is already the newest version (3.22.1-1ubuntu1.22.04.2).
> file is already the newest version (1:5.41-3ubuntu0.1).
> gdb is already the newest version (12.1-0ubuntu1~22.04.2).
> git is already the newest version (1:2.34.1-1ubuntu1.12).
> libxml2-dev is already the newest version (2.9.13+dfsg-1ubuntu0.7).
> libxml2-utils is already the newest version (2.9.13+dfsg-1ubuntu0.7).
> python3 is already the newest version (3.10.6-1~22.04.1).
> python3-dev is already the newest version (3.10.6-1~22.04.1).
> python3-setuptools is already the newest version (59.6.0-1.2ubuntu0.22.04.2).
> rsync is already the newest version (3.2.7-0ubuntu0.22.04.4).
> unzip is already the newest version (6.0-26ubuntu3.2).
> python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.5).
> python3-wheel is already the newest version (0.37.1-2ubuntu0.22.04.1).
> The following packages were automatically installed and are no longer required:
>   usb-modeswitch usb-modeswitch-data
> Use 'sudo apt autoremove' to remove them.
> 0 upgraded, 0 newly installed, 0 to remove and 266 not upgraded.
> 
> Installing PX4 Python3 dependencies
> Requirement already satisfied: argcomplete in /usr/lib/python3/dist-packages (from -r /home/robotica/PX4-Autopilot/Tools/setup/requirements.txt (line 1)) (1.8.1)
> Collecting argparse>=1.2
>   Using cached argparse-1.4.0-py2.py3-none-any.whl (23 kB)
> Collecting cerberus
>   Using cached Cerberus-1.3.7-py3-none-any.whl (30 kB)
> Collecting coverage
>   Using cached coverage-7.8.0-cp310-cp310-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (239 kB)
> Collecting empy==3.3.4
>   Using cached empy-3.3.4.tar.gz (62 kB)
>   Preparing metadata (setup.py) ... error
>   error: subprocess-exited-with-error
>   
>   × python setup.py egg_info did not run successfully.
>   │ exit code: 1
>   ╰─> [30 lines of output]
>       Traceback (most recent call last):
>         File "<string>", line 2, in <module>
>         File "<pip-setuptools-caller>", line 34, in <module>
>         File "/tmp/pip-install-_wmn9ol3/empy_833662e891c94094b9bebd4ca1cdf531/setup.py", line 24, in <module>
>           setup(
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 186, in setup
>           return run_commands(dist)
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 202, in run_commands
>           dist.run_commands()
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 1002, in run_commands
>           self.run_command(cmd)
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/dist.py", line 1104, in run_command
>           super().run_command(command)
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 1021, in run_command
>           cmd_obj.run()
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 312, in run
>           self.find_sources()
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 320, in find_sources
>           mm.run()
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 548, in run
>           self.prune_file_list()
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/command/sdist.py", line 162, in prune_file_list
>           super().prune_file_list()
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/_distutils/command/sdist.py", line 386, in prune_file_list
>           base_dir = self.distribution.get_fullname()
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/_core_metadata.py", line 275, in get_fullname
>           return _distribution_fullname(self.get_name(), self.get_version())
>         File "/home/robotica/.local/lib/python3.10/site-packages/setuptools/_core_metadata.py", line 293, in _distribution_fullname
>           canonicalize_version(version, strip_trailing_zero=False),
>       TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'
>       [end of output]
>   
>   note: This error originates from a subprocess, and is likely not a problem with pip.
> error: metadata-generation-failed
> 
> × Encountered error while generating package metadata.
> ╰─> See above for output.
> 
> note: This is an issue with the package mentioned above, not pip.
> hint: See above for details.

### To Reproduce

1: enter the PX4 folder in terminal
2: run the ubuntu.sh script

### Vehicle type

None

### Additional context

This happens when trying to setup PX4 itself

MoonlightShinesparkle, 2025-05-20:
@mrpollo Here is the bug report you told me to generate, hopefully we can track the source soon enough

awaisk65, 2025-08-19:
i guess the error you're encountering when running the PX4 setup script on Ubuntu 22.04 is related to a conflict between the empy==3.3.4 package and a newer version of setuptools installed in your Python environment. May be downgrading setuptools solve the issue