PX4/PX4-Autopilot #24898: Add PX4IO serial support (px4io_serial.cpp/.h) for NXP K66-based flight controllers (e.g. fmuk66-v3)

sahil-enercomp, 2025-05-21:
### Describe problem solved by the proposed feature

We are using the NXP K66 (MK66FX1M0) as the FMU in a custom PX4-based flight controller (fmuk66-v3). The board has a limited number of PWM outputs available through its native pins, which becomes a bottleneck when we need to control more actuators (e.g. in VTOL or complex multirotor configurations).

To extend PWM outputs, we are trying to integrate an external I/O co-processor using an STM32F103 ("Blue Pill") running the PX4IO firmware (px4io-v2_default). This mimics the I/O architecture found in Pixhawk 2.x systems, where the FMU communicates with the I/O via UART.

However, the PX4 codebase does not currently include px4io_serial.cpp, px4io_serial.h, or the corresponding CMakeLists.txt under:

**_**_platforms/nuttx/src/px4/nxp/k66/_**_**

Other platforms like STM32F4, F7, H7, and even some NXP variants include these files, which are required to enable PX4IO UART communication (PX4IO_SERIAL interface).

Without these files, the PX4IO driver cannot be used on K66-based boards even if the hardware and configuration support it.

### Describe your preferred solution

We would like official support for PX4IO UART communication (px4io_serial) to be added for NXP K66-based boards, specifically fmuk66-v3.

This includes:

-     Implementing or porting px4io_serial.cpp and px4io_serial.h under platforms/nuttx/src/px4/nxp/k66/
-     Adding the corresponding CMakeLists.txt to compile them
-     Ensuring proper macro definitions (CONFIG_ARCH_BOARD_FMUK66_V3) are present for conditional compilation

We have drafted a version of these files based on other board implementations and are attaching them here for review.

### Describe possible alternatives

Rewriting the PX4IO communication entirely on the FMU side as a custom PX4 module (outside standard px4io_serial flow) — but this breaks compatibility with the existing PX4IO firmware and would require duplicating a lot of logic.

### Additional context

We believe this enhancement is a minimal, low-risk code addition that would significantly improve support for the NXP K66 family in PX4 and enable a broader set of use cases.

We are attaching our implementation of:

-     px4io_serial.cpp
-     px4io_serial.h
-     CMakeLists.txt

These were adapted from other NuttX-based platforms in PX4 and adjusted to match the FMUK66 hardware layout. However, we are not certain if the implementation fully adheres to all necessary PX4 conventions and would greatly appreciate review and integration by the core PX4 team.

[CMakeLists.txt](https://github.com/user-attachments/files/20369275/CMakeLists.txt)

[px4io_serial.cpp.txt](https://github.com/user-attachments/files/20369389/px4io_serial.cpp.txt)

[px4io_serial.h.txt](https://github.com/user-attachments/files/20369400/px4io_serial.h.txt)

PetervdPerk-NXP, 2025-05-23:
Hi @sahil-enercomp thank your for your contribution.

The fmuk66-v3 never had an PX4IO thus it was never implemented, the newer IMXRT1176 based FMU does support PX4IO, But also has more onboard PWM's (12 vs 8) if needed.
Overall I think it looks okay, but best would be to do a pull-request with the changes.

PetervdPerk-NXP, 2025-05-27:
Hi Sahil,

There's an IMXRT1062 Teensy-based VMU which is low-cost, but also a bit I/O limited due to Teensy itself.
https://github.com/NXPHoverGames/TropicCommunityVMU

Furthermore there's also a IMXRT1064 based solution in the works that looks like the Teensy one but has more I/O available.
https://github.com/PX4/PX4-Autopilot/tree/pr-mr-tropic

closed 2026-01-12 by github-actions[bot]