PX4/PX4-Autopilot #24903: Forward VTX CRSF control packets to VTX UART

jsm09a, 2025-05-22:
### Describe problem solved by the proposed feature

Some VTX units (e.g. TBS Unify Pro32 HV) can be administered via CrossFire packets.  This allows remote adjustment of Frequency, Power Level, and Pit Mode on the Video Transmitter: [https://github.com/tbs-fpv/tbs-crsf-spec/blob/main/crsf.md#0x320x08-vtx-commands](url).

Currently, CrsfRc.cpp silently drops such packets.  It would be helpful if they were routed to the UART connected to the VTX.  This should allow VTX administration directly from applicable RC units (e.g. RadioMaster Boxer) via their ELRS LUA script.

### Describe your preferred solution

See [https://discord.com/channels/1022170275984457759/1375147454940319866/1375201510278828162](url) for a suggested implementation by "dakejahl".

### Describe possible alternatives

I am not familiar with the PX4 architecture and the preferred way of doing things ... but an alternative might be to just "mark" a UART port as a "CRSF Mirror" and then simply copy all received bytes to the output FIFO for that port as per [https://github.com/tbs-fpv/tbs-crsf-spec/blob/main/crsf.md#routing](url).

dk7xe, 2025-06-11:
It's in main as of now. See https://github.com/PX4/PX4-Autopilot/pull/24982

niklaut, 2025-10-02:
Is there a protocol translation being done, or is it just literally copying data around? Cos in the latter case you could create a Y-cable that routes the CRSF data to PX4 RX and the VTX RX.