PX4/PX4-Autopilot #24925: [Bug] 3DR control zero h7 oem rev g issue with MTD driver

dirksavage88, 2025-05-27:
### Describe the bug

```
HW arch: 3DR_CTRL_ZERO_H7_OEM_REVG
PX4 git-hash: e90b0b6bb912e9d367932498cfed9ee604b5eab3
PX4 version: 1.15.0 80 (17760384)
PX4 git-branch: prerelease-3dr-m10059g
OS: NuttX
OS version: Release 12.6.0 (201720063)
OS git-hash: 7b0bc80b53063eb3fab7569d0268a19152ceb421
Build datetime: Aug  9 2024 20:37:11
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 0006000000003339393233325118001d0022
MCU: STM32H7[4|5]xxx, rev. V
ERROR [bsondump] open '/fs/mtd_caldata' failed (2)
New /fs/mtd_caldata size is:
ERROR [bsondump] open '/fs/mtd_caldata' failed (2)
INFO  [param] selected parameter default file /fs/mtd_params
ERROR [param] open '/fs/mtd_params' failed (2)
ERROR [init] param import failed
ERROR [bsondump] open '/fs/mtd_params' failed (2)
INFO  [parameters] summary: 0/1962 (used/total)
INFO  [parameters] file: /fs/mtd_params
INFO  [parameters] storage array: 0/1962 elements (256 bytes total)
INFO  [parameters] storage array (custom defaults): 0/1962 elements (256 bytes total)
INFO  [parameters] auto save: on
param: export: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
param: find: 0 events
param: get: 0 events
param: set: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
dmesg [46:100]
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
ERROR [PX4_MTD] failed to initialize mtd driver
ERROR [PX4_MTD] mtd failure: -5 bus 2 address 0 class 1
HW arch: 3DR_CTRL_ZERO_H7_OEM_REVG
PX4 git-hash: e90b0b6bb912e9d367932498cfed9ee604b5eab3
PX4 version: 1.15.0 80 (17760384)
PX4 git-branch: prerelease-3dr-m10059g
OS: NuttX
OS version: Release 12.6.0 (201720063)
OS git-hash: 7b0bc80b53063eb3fab7569d0268a19152ceb421
Build datetime: Aug  9 2024 20:37:11
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 0006000000003339393233325118001d0022
MCU: STM32H7[4|5]xxx, rev. V
ERROR [bsondump] open '/fs/mtd_caldata' failed (2)
New /fs/mtd_caldata size is:
ERROR [bsondump] open '/fs/mtd_caldata' failed (2)
INFO  [param] selected parameter default file /fs/mtd_params
ERROR [param] open '/fs/mtd_params' failed (2)
ERROR [init] param import failed
ERROR [bsondump] open '/fs/mtd_params' failed (2)
INFO  [parameters] summary: 0/1962 (used/total)
INFO  [parameters] file: /fs/mtd_params
INFO  [parameters] storage array: 0/1962 elements (256 bytes total)
INFO  [parameters] storage array (custom defaults): 0/1962 elements (256 bytes total)
INFO  [parameters] auto save: on
param: export: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
param: find: 0 events
param: get: 0 events
param: set: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
dmesg [46:100]
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaultsERROR [PX4_MTD] failed to initialize mtd driver
ERROR [PX4_MTD] mtd failure: -5 bus 2 address 0 class 1
HW arch: 3DR_CTRL_ZERO_H7_OEM_REVG
PX4 git-hash: e90b0b6bb912e9d367932498cfed9ee604b5eab3
PX4 version: 1.15.0 80 (17760384)
PX4 git-branch: prerelease-3dr-m10059g
OS: NuttX
OS version: Release 12.6.0 (201720063)
OS git-hash: 7b0bc80b53063eb3fab7569d0268a19152ceb421
Build datetime: Aug  9 2024 20:37:11
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 0006000000003339393233325118001d0022
MCU: STM32H7[4|5]xxx, rev. V
ERROR [bsondump] open '/fs/mtd_caldata' failed (2)
New /fs/mtd_caldata size is:
ERROR [bsondump] open '/fs/mtd_caldata' failed (2)
INFO  [param] selected parameter default file /fs/mtd_params
ERROR [param] open '/fs/mtd_params' failed (2)
ERROR [init] param import failed
ERROR [bsondump] open '/fs/mtd_params' failed (2)
INFO  [parameters] summary: 0/1962 (used/total)
INFO  [parameters] file: /fs/mtd_params
INFO  [parameters] storage array: 0/1962 elements (256 bytes total)
INFO  [parameters] storage array (custom defaults): 0/1962 elements (256 bytes total)
INFO  [parameters] auto save: on
param: export: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
param: find: 0 events
param: get: 0 events
param: set: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
dmesg [46:100]
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
ERROR [PX4_MTD] failed to initialize mtd driver
ERROR [PX4_MTD] mtd failure: -5 bus 2 address 0 class 1
HW arch: 3DR_CTRL_ZERO_H7_OEM_REVG
PX4 git-hash: e90b0b6bb912e9d367932498cfed9ee604b5eab3
PX4 version: 1.15.0 80 (17760384)
PX4 git-branch: prerelease-3dr-m10059g
OS: NuttX
OS version: Release 12.6.0 (201720063)
OS git-hash: 7b0bc80b53063eb3fab7569d0268a19152ceb421
Build datetime: Aug  9 2024 20:37:11
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 0006000000003339393233325118001d0022
MCU: STM32H7[4|5]xxx, rev. V
ERROR [bsondump] open '/fs/mtd_caldata' failed (2)
New /fs/mtd_caldata size is:
ERROR [bsondump] open '/fs/mtd_caldata' failed (2)
INFO  [param] selected parameter default file /fs/mtd_params
ERROR [param] open '/fs/mtd_params' failed (2)
ERROR [init] param import failed
ERROR [bsondump] open '/fs/mtd_params' failed (2)
INFO  [parameters] summary: 0/1962 (used/total)
INFO  [parameters] file: /fs/mtd_params
INFO  [parameters] storage array: 0/1962 elements (256 bytes total)
INFO  [parameters] storage array (custom defaults): 0/1962 elements (256 bytes total)
INFO  [parameters] auto save: on
param: export: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
param: find: 0 events
param: get: 0 events
param: set: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
dmesg [46:100]
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
nsh> 
```

### To Reproduce

1. Flash the un-official build release for the new revg
2. Open up QGC and attempt to select an airframe
3. Parameters do not write

### Expected behavior

MTD driver should initialize and paramter read/write works

### Software Version

Custom version of PX4 for the pre-release of the 3dr control zero h7 oem revg

### Flight controller

3dr control zero h7 oem revg

### Vehicle type

Fixed Wing

### Additional context

@TOTON95

TOTON95, 2025-11-19:
@dirksavage88 I could find a solution, and I can recreate it in new-made and used boards. I will submit the PRs in the morning  🙌🏼

TOTON95, 2025-11-19:
Currently,  waiting for this PR to be merged https://github.com/PX4/NuttX/pull/357

closed 2025-11-22 by dirksavage88