PX4/PX4-Autopilot #24926: [Docs] List DDS topics reference

mrpollo, 2025-05-27:
### Describe problem solved by the proposed feature

The list of topics available trough the uXRCE-DDS interface is defined in a dds_topics.yml file and the relevant docs page only shows an example of the structure, leaving users to read the source to find the list. This was enough since the list wasn't offering much data, and the details for each uORB message could be found on the uORB Message Reference. However with the introduction of the optional `rate_limit` field, the DDS Topics YAML is now home to unique data, it's likely better if we have a message list rendered on its own page in the docs that has direct links to the uORB message reference, and also displays the `rate_limit` and any other future fields.

### Describe your preferred solution

I think we should finally add a list of messages in the dds_topics.yml to our docs, perhaps as a pretty table, with three columns similar to the example below. We should also automate the generation of markdown similar to what happens with uORB message reference.

# List of Topics
## Rate Limit
An additional (optional) `rate_limit` field (only for publication entries), which specifies the maximum rate (Hz) at which messages will be published on this topic by PX4 to ROS 2. If left unspecified, the maximum publication rate limit is set to 100 Hz.

## Messages
### Publications
| Topic | Type | Rate Limit |
| --- | --- | --- |
| /fmu/out/battery_status | px4_msgs::msg::BatteryStatus | 1 |
| /fmu/out/timesync_status | px4_msgs::msg::TimesyncStatus | 10 |

### Subscriptions
| Topic | Type |
| --- | --- |
| /fmu/in/distance_sensor | px4_msgs::msg::DistanceSensor |
| /fmu/in/offboard_control_mode | px4_msgs::msg::OffboardControlMode |

### Describe possible alternatives

keep it the way we have it right now

mrpollo, 2025-05-27:
@GuillaumeLaine I saw your work on #24765 and #24924, good stuff! What do you think of this proposal?

cc @hamishwillee

GuillaumeLaine, 2025-05-27:
@mrpollo i think it makes sense to have an exhaustive list of topics and their dds publication rates documented 👍🏻

hamishwillee, 2025-05-28:
Yes, this is a very good idea.
I'll write a parser to extract this.

closed 2025-06-05 by mrpollo

hamishwillee, 2025-06-11:
Note, #25008 plugs the generated markdown version of the yaml file. into the docs so it is linked and cross referenced (the other PR updated the docs to generate the file).