PX4/PX4-Autopilot #24969: [Bug] Entering OFFBOARD mode before takeoff causes unexpected error

wyunc, 2025-06-04:
### Describe the bug

Hello,
This issue has existed for quite a long time and is similar to #22919. If I directly switch the drone in jmavsim to 'OFFBOARD' mode before takeoff, it results in the following error:
`ERROR [simulator_mavlink] poll timeout 0, 22`

However, if I first load a mission, switch to 'AUTO.MISSION' mode, let it run, and then switch to 'OFFBOARD' mode, the problem does not occur.

I found [relevant documentation ](https://docs.px4.io/main/en/flight_modes/offboard.html) for 'OFFBOARD' mode here: 
"PX4 must receive a stream of MAVLink setpoint messages or the ROS 2 OffboardControlMode at 2 Hz as proof that the external controller is healthy. The stream must be sent for at least a second before PX4 will arm in offboard mode, or switch to offboard mode when flying. If the rate falls below 2Hz while under external control PX4 will switch out of offboard mode after a timeout (COM_OF_LOSS_T), and attempt to land or perform some other failsafe action. The action depends on whether or not RC control is available, and is defined in the parameter COM_OBL_RC_ACT."

However, I noticed that even when setting [COM_OF_LOSS_T ](https://docs.px4.io/main/en/flight_modes/offboard.html#COM_OF_LOSS_T)= 10 seconds (which, according to the docs, should trigger failsafe only after a 10-second timeout), the parameter seems ineffective — jmavsim still enters the error state after a short delay (around 1 second).

Additionally, based on the documentation, after the timeout, the system should switch to the mode defined by [COM_OBL_RC_ACT ](https://docs.px4.io/main/en/flight_modes/offboard.html#COM_OBL_RC_ACT)(0: Position, 1: Altitude, 2: Manual, 3: Return, 4: Land). However, this switch does not actually happen — jmavsim simply loses connection.

In short, I think the current documentation does not clearly explain why jmavsim loses connection in this case. This seems to be an unintended behavior.

### Software Version

1.16.0

### Vehicle type

Multicopter

feomey, 2025-07-04:
Did you solve this issue? I'm encountering the same problem

feomey, 2025-07-05:
Hi@wyunc ,

"ERROR [simulator_mavlink] poll timeout 0, 22" was resolved by switching to PX4 v1.12.3, and I've set the data transmission rate at 2Hz. The current issue is: the drone can successfully arm, but fails to switch to OFFBOARD mode. By using 'rostopic echo /mavros/state', I can see the mode remains 'AUTO.LOITER'.

Later, in the PX4 simulation program, I entered the command commander mode offboard to forcibly switch modes, and it displayed the same messages as when I used the offboard code:
'INFO [tone_alarm] notify negative' and
'[WARN] [1751734333.958833334, 426.056000000]: CMD: Unexpected command 176, result 1'.
After researching, I found that this issue is typically caused by ​not continuously sending position setpoint data. However, my code is based on the ​official offboard example, which ​does include continuous position setpoint transmission.

Below is my code：

[image]

[image]

Here are some node statuses, information, and error messages
[image]

[image]

Thank you for your detailed explanation. I apologize in advance if this has caused any inconvenience.

feomey, 2025-07-06:
I have resolved the issue. The inability to enter Offboard mode was caused by the drone lacking positioning data. My flight control simulation program is modified based on PX4 and differs slightly from the standard PX4 control program. I needed to send position data to the mavros/setpoint_raw/local node to enable arming.