PX4/PX4-Autopilot #25057: [Bug] MultiRotor Delay Until

MasonPhillip, 2025-06-16:
### Describe the bug

Upon reaching a delay, the multirotor "paces" back and forth until the delay ends, rather than staying still

### To Reproduce

Connected to gazebo
Uploaded mission and parameters
Took off
At mission item 2/3 the bug is visible

[mission file.txt](https://github.com/user-attachments/files/20763902/mission.file.txt)

[parameters.txt](https://github.com/user-attachments/files/20763903/parameters.txt)

### Expected behavior

I would expect it to sit at the delay or at the acceptance radius of the delay

### Screenshot / Media

[Video](https://drive.google.com/file/d/1dgqTOMKG_tgPpLJirEvbhHmr2cr8fA8d/view?usp=sharing)

### Flight Log

[Flight Review](https://review.px4.io/plot_app?log=ccb16003-f996-46aa-945f-db142ad681f5)

### Software Version

1.15

### Flight controller

https://docs.px4.io/main/en/sim_gazebo_gz/vehicles.html#x500-quadrotor

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

No wiring, vehicle simulated with Gazebo
https://docs.px4.io/main/en/sim_gazebo_gz/vehicles.html#x500-quadrotor

### Additional context

I tried to upload the mission file and parameters as .kml,  and .mission/.params,  but the only file type it would accept was .txt

MasonPhillip, 2025-06-23:
If the point after the delay is a land, it behaves how I would expect, but if the next point is a waypoint, the issue is present

dakejahl, 2025-06-23:
@MasonPhillip does it occur on the main branch? Is it reproducible in simulation?

MasonPhillip, 2025-06-23:
@dakejahl Yes it occurs on the main branch and I encountered it using the Gazebo Simulation

closed 2025-10-15 by dakejahl