PX4/PX4-Autopilot #25093: [Bug] Optical Flow doesn’t work on 1.16.x (latest)

lida2003, 2025-06-22:
### Describe the bug

[Optical Flow doesn’t work on 1.16.x (latest)](https://discuss.px4.io/t/optical-flow-doesnt-work-on-1-16-x-latest/46029)

### Vehicle type

None

lida2003, 2025-06-28:
@DronecodeBot ??? robot?

lida2003, 2025-07-06:
@dakejahl I have switched back to 1.15.4, which works fine. I'll check the log when I have time to test latest 1.16.x release.

dakejahl, 2025-12-12:
I've got one of these sensors in hand now. I just need to find some time to test it.