PX4/PX4-Autopilot #25118: [Bug] set param MAV_{i}_BROADCAST = 1 to enable network

Tony-Dja, 2025-06-25:
### Describe the bug

Hello,

I can't broadcast mavlink outside my PX4 Docker container.

In the logs, I have
INFO [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)

I tried modifying the px4-rc.mavlink file by adding => param set MAV_0_BROADCAST 1.

But the parameter is never taken into account.

I want to broadcast mavlink to another container containing ROS2 and mavros on port 14540. It's listening on the port, but I can't broadcast to it from the PX4 container.

I'm on a host network, with PX4 1.15.4 and ROS2 jazzy.

Do you know where i can enable broadcast ?

Thanks

### Vehicle type

None

julianoes, 2025-07-24:
Check how it's done here:
https://github.com/JonasVautherin/px4-gazebo-headless

andrewjong, 2025-09-22:
Might anyone have specific instructions on how to make this work, please? I'm a bit confused. Thanks!