PX4/PX4-Autopilot #25143: [FEATURE] Gazebo World with disabled physics engine for use as visualizer

hamishwillee, 2025-07-02:
### Describe problem solved by the proposed feature

The SIH simulator uses JMAVsim for a visualizer, but JMAVSim is now deprecated and unreliable. 

According to @beniaminopozzan ([see here](https://github.com/PX4/PX4-Autopilot/pull/25095#discussion_r2173479186)) it is possible for Gazebo to allows this, but it requires

> a world with disabled physics were the drone position flows in the reverse direction. Instead simulator -> px4 it needs to be px4 -> simulator.

This would then allow Gazebo to act as simple rendering software.

SIH is very useful, so I think this is a real need.

### Describe your preferred solution

As pre description

### Describe possible alternatives

Another out of the box visualizer that we could use.

It seems to me that Gazebo is the platform where we are investing though, so good to have it do everything.

hamishwillee, 2025-07-02:
@beniaminopozzan @Jaeyoung-Lim I know you are both busy, but can you recommend how we might make this happen?

From a docs perspective the follow on would be an update to https://docs.px4.io/main/en/sim_sih/#display-visualisation-optional

Jaeyoung-Lim, 2025-07-02:
@hamishwillee I have been thinking of something similar, but I think using gazebo diminishes the value of SIH: if you can run gazebo, you can run gazebo

What would you want to visualize? We could also think of writing a small 3D visualizer for SIH with something like open3D or cesium

hamishwillee, 2025-07-02:
> My question is rather what is the problem of jmavsim? why did it get deprecated?

PX4 isn't willing to invest anything in ongoing maintenance - "all resources on Gazebo". It now doesn't reliably work.

hamishwillee, 2025-07-09:
So where does that leave us?

Jaeyoung-Lim, 2025-07-09:
@hamishwillee I think we need to find more people to maintain the stack. I hope the Dronecode foundation would actively looks for additional people to fix this problem.

If you feel like SIH is not working and not enough maintanance is being done, my suggestion would be to remove it completely from the PX4 stack. I don't see any value of SIH, if we are to use gazebo as a visualizer. It does not complement anything that is missing on the gazebo side.

Maybe @dakejahl can probably chime in as well

dakejahl, 2025-07-09:
> I have no strong feelings about anything other than the sanity of the docs 

We should make you a shirt xD

dakejahl, 2025-07-09:
If someone invested a little bit of time I bet it wouldn't be too hard to vibe code a simple visualizer. Agree with Jae about maintenance, we need more people to help out across the board

hamishwillee, 2025-07-09:
@dakejahl A shirt would be nice :-). 

FYI @bresch Uses it, but without the visualizer.

> it's simpler for me to adjust the simulation models. For example, I changed the world to be ellipsoidal (instead of flat) to be able to test the behavior of the EKF over long distances, also, it's easy to change the dynamic response of a drone, add actuators/control surfaces without having to mess up with/recompile Gazebo (edited)

Jaeyoung-Lim, 2025-07-10:
@hamishwillee I can try and write a simple visualizer based on python

sfuhrer, 2025-07-10:
We use SIH (developers) and have plans to [replace the current gazebo-classic simulations in the CI with SIH](https://github.com/PX4/PX4-Autopilot/pull/24237). SIH running on HW is also a thing and we are without alternatives there.
While a simple visualizer would be nice in some cases it's not a must-have IMO.

MaEtUgR, 2025-12-03:
> SIH is too perfect in some cases.

It rather has too much noise than being too perfect but let's improve that. CI on SIH in progress: https://github.com/PX4/PX4-Autopilot/pull/26032