PX4/PX4-Autopilot #25188: [Bug] XRCE_DDS Bridge does not work in SITL across docker containers because PX4_PARAM_UXRCE_DDS_AG_IP is not respected

sansha, 2025-07-10:
### Describe the bug

Hey, 

I am having issues setting up SITL with ROS2 with two seperate docker containers on an isolated network. I found a fix by modifying a config file and think it is a bug that `PX4_PARAM_UXRCE_DDS_AG_IP` is not respected.

## What do I want?

Two docker containers

- `offboard` running ros2 and the xrce agent  that is exactly the same as running on the drone
- `px4_sim` that simulates PX4 and the XRCE_DDS bridge to ROS2

running in an isolated network on docker defined with 

```
networks:
  isolated_network:
    driver: bridge
    ipam:
      config:
        - subnet: 172.40.0.0/16
          gateway: 172.40.0.1
```

When using the `host` network, everything works.

### To Reproduce

## What should work but doesn't 

Using that docker network, I 
- set the IP of `offboard` to `ipv4_address: 172.40.0.3` 
-  configure `px4_sim` to have environment variable `PX4_PARAM_UXRCE_DDS_AG_IP=-1406664701` (matching the IP Address in INT32)

I can see that environment variable being set correctly in Qgroundcontrol.

However, the bridge still does not work.

## The fix

In this file 
https://github.com/PX4/PX4-Autopilot/blob/14c0908082b8a9135895e1928e04b4775338ab54/ROMFS/px4fmu_common/init.d-posix/rcS#L322C1-L322C76

the uxrce_dds_client is started with a hardcoded IP address that overrides any environment variable.
```
uxrce_dds_client start -t udp -h 127.0.0.1 -p $uxrce_dds_port $uxrce_dds_ns
```

if I modify this line on my docker container to be 
```
uxrce_dds_client start -t udp  -p $uxrce_dds_port $uxrce_dds_ns
```
and restart everything, the parameter is accepted, the correct IP is set and the bridge works.

### Software Version

I am not sure but saw that this line exists in all latest versions and `main`.

### Vehicle type

Multicopter

### Additional context

So I made it work for now by modifying that file in my docker build but it seems hacky to modify the config file and I think this is a bug and instead 
- this config file should not start the client with a fixed IP
- to not break existing setups, one could set a default param to be `127.0.0.1` 

Thank you for your consideration. I am still new to PX4 so I hope I did not miss anything.

Also happy to make a PR :)

sansha, 2025-07-16:
Created a PR: #25231

closed 2025-07-25 by beniaminopozzan

alexespinoza28, 2025-08-07:
When I tried this fix it still didn't work. Instead of erasing ' -h 127.0.0.1' I replaced the ip with the inet ip from this command 'ip addr show eth0 | grep inet' on wsl and my sitl running on docker in wsl and XRCE_DDS on wsl connected instantly.

PR #25231 (2025-07-25): Fix issue where XRCE_DDS Bridge did not respect environment variables

### Solved Problem

Fixes #25188 (detailed description in that issue)

### Solution

Hi, this is the PR fixing #25188 by removing the `-h 127.0.0.1` flag in the launch command for the client.
This fix works for me, not sure if other setups are affected.

Let me know your thoughts!  

### Changelog Entry
For release notes:
```
Feature/Bugfix Fix bug where UXRCE_DDS_AG_IP was not read when launching the xrce client in SITL.
```

### Alternatives

In the first commit I load the env variable explicitly but changed this to just  [... cut]