PX4/PX4-Autopilot #25237: [Docs] [Bug] OSD is not documented

hamishwillee, 2025-07-17:
### Describe the bug

Just saw #25233, which requests a full implementation.

Before we do that, it would be great to have any documentation on what is supported w.r.t. OSD.

All we have now is 
- information on setting the OSD mavlink stream in https://docs.px4.io/main/en/peripherals/mavlink_peripherals.html
- OSD parameters https://docs.px4.io/main/en/advanced_config/parameter_reference.html#osd
- Module with no useful information https://docs.px4.io/main/en/modules/modules_driver.html#msp-osd
- Reptile dragon info https://docs.px4.io/main/en/frames_plane/reptile_dragon_2.html#msp-osd-module

What we need is a page on MSP OSD - what is it? How does it differ from other OSD? What modules support this "whatever it is"? How do you wire them into PX4?

> [!WARNING] 
> "an undocumented feature does not exist".

SolderSyntax, 2025-07-17:
Should we have a dedicated OSD page under docs.px4.io/main/peripherals?

I think most of the OSD parameters are clear enough, although I'm not completely sure all of them are being used.

The actual setup of OSD depends on the type. MSP OSD uses serial communication over any uart, so connection wise it's the same. Setting it up in PX4 is pretty straight forward with the module's parameters.

As for the MSP OSD module - I have to spend some time getting familiar with before I can add anything (code or documentation).

Mavlink OSD stream mode is not configurable - it's hard coded as seen [here](https://github.com/PX4/PX4-Autopilot/blob/78a2a3d0dd244e9604384e0b5eabd8c0c3bcd313/src/modules/mavlink/mavlink_main.cpp#L1618). If we'll have an OSD page for PX4 we can mention it over there, as an option. I also don't have any example for devices that implement OSD with Mavlink data.

SolderSyntax, 2025-07-23:
Yes, I can get something going.
I'd have to get myself familiar with how the Docs are written and built.

hamishwillee, 2025-07-23:
@SolderSyntax There are some guidelines here https://docs.px4.io/main/en/contribute/docs.html but actually it is quite easy.

1. Docs are in the PX4 source tree - they are source code in markdown under `/docs/en/<somefolder>`
2. So you create files just like source. In this case you might copy the optical flow markdown file into /en/perhipherals as a starting point.
3. To make the doc appear in the sidebar you add a link in the pattern in docs/en/SUMMARY.md

You can test building by installing node - greater than version 22 (I use nvm to install node versions as needed). Then install `yarn` globally (node install yarn -g`, then go to the /docs/ directory and enter `yarn start`. That's it. Otherwise it is just code.

hamishwillee, 2025-12-17:
@SolderSyntax Do you think you might be able to look at this?