PX4/PX4-Autopilot #25248: [Bug] OpticalFlow Local Position Estimation doesn't work well on 1.16

F00dP00l, 2025-07-18:
### Describe the bug

I recently built a indoor self-navigation multicopter with **opticial flow sensor and tof distance sensor**, but I found it impossible to get the drone into Position mode with whatever configuration I tried. Since I use a Mavlink optical flow sensor and its output seems quiet well, I am confident that _there's nothing to do with my sensors_. I will leave my finnal config below in case any of you wants to estimate it.

[2025-07-18.zip](https://github.com/user-attachments/files/21313958/2025-07-18.zip)

### Vehicle type

None

F00dP00l, 2025-07-18:
I tried a older version of PX4(in this case, 1.14). It worked just fine. I can tell there is certainly something wrong going on with version 1.16.

w2016561536, 2025-07-23:
Build from v1.16-rc2，this bug still. However it works fine with a GPS or external vision system.

DRT-Jin, 2025-09-29:
I have the same issue. I tried building with PX4 v1.16 stable and v1.16-rc2 (and other tags), but all give the same result. Very rarely the position mode stays active, but in most flights it switches to altitude mode right after takeoff. PX4 v1.15.4 and similar firmware versions have no such problem.

franciscoxavier04, 2025-10-15:
I have the same issue using the MicroAir743v2 with MTF-01. I'm running PX4 v1.16 RC3.

As soon as I take off, I get these msgs:

[23:41:45.555] Info: Takeoff detected
[23:41:49.795] Critical: Failsafe activated: Autopilot disengaged, switching to Altitude mode
Navigation error: No valid position estimate

EXPX3, 2026-01-09:
Also have issues with ARK Flow based Dexi Droneblocks UAV with PX4 1.16 version. 
Especially in offboard mode we were not able to make a simple takeoff to 2m and landing in indoors.
What worked mostly was flying the drone in position, altitude and manual mode, that too only if we take off quick enough above the ground. 
We observed odometry estimation "sensor_combined" shooting off very quick especially in x,y directions when manually moving the drone from a place to another which is 1 m away without arming.

@dakejahl could you please elaborate on what basis you said "ARK Flow works without any issues" in the above comment of yours?

Thankyou.