PX4/PX4-Autopilot #25271: [Bug] Can't start Gazebo simulator => error: unknown target 'gz_x500'

GibbekAdsk, 2025-07-23:
### Describe the bug

I followed the steps to install PX4-Autopilot described [here](https://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html) and [in this video](https://www.youtube.com/watch?v=sZbJOSk8Cc4).
I installed Ubuntu 22.04.5 LTS on WSL2 (Win11), I ran the script `bash ./PX4-Autopilot/Tools/setup/ubuntu.sh` and I was able to use/run `make px4_sitl jmavsim` but I was not able to get the Gazebo simulator running.

In the PX4-Autopilot subfolder I ran...
`make distclean`
`make clean`
and then 
`make px4_sitl gz_x500`
but I get the error message: `ninja: error: unknown target 'gz_x500'`

This is the output of command: `make px4_sitl gz_x500`
_-- PX4_GIT_TAG: v1.16.0-rc1-447-gf0ecd9e757
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3")
-- PX4 config file: /home/gibbek/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- TESTING y
-- ETHERNET y
-- ROOT_PATH .
-- PARAM_FILE /fs/mtd_params
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is GNU 11.4.0
-- The C compiler identification is GNU 11.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: RelWithDebInfo
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Looking for gz-transport13 -- found version 13.4.1
-- Searching for dependencies of gz-transport13
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found version "3.12.4")
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.4
-- Found ZeroMQ: TRUE (Required is at least version "4")
-- Checking for module 'uuid'
--   Found uuid, version 2.37.2
-- Found UUID: TRUE
-- Looking for gz-utils2 -- found version 2.2.1
-- Searching for dependencies of gz-utils2
-- Searching for <gz-utils2> component [cli]
-- Looking for gz-utils2-cli -- found version 2.2.1
-- Searching for dependencies of gz-utils2-cli
-- Looking for gz-msgs10 -- found version 10.3.2
-- Searching for dependencies of gz-msgs10
-- Looking for gz-utils2 -- found version 2.2.1
-- Looking for gz-math7 -- found version 7.5.1
-- Searching for dependencies of gz-math7
-- Looking for gz-utils2 -- found version 2.2.1
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Looking for gz-transport13 -- found version 13.4.1
-- Searching for dependencies of gz-transport13
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.4
[... 246 lines cut ...]
-- Found OpenCV: /usr (found version "4.5.4") found components: core imgproc
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2")
-- Checking for module 'gstreamer-1.0'
--   Found gstreamer-1.0, version 1.20.3
-- Checking for module 'gstreamer-app-1.0'
--   Found gstreamer-app-1.0, version 1.20.1
-- Found Java: /usr/bin/java (found version "11.0.27")
-- ROMFS: ROMFS/px4fmu_common
Architecture:  amd64
==> CPACK_INSTALL_PREFIX = @DEB_INSTALL_PREFIX@
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gibbek/PX4-Autopilot/build/px4_sitl_default
ninja: error: unknown target 'gz_x500'
make: *** [Makefile:227: px4_sitl] Error 1_

Following version(s) of Gazebo are installed:
`gz sim --versions`  => 8.9.0
`gz gui --versions`  => 8.4.0

Any advice?

### Vehicle type

None

dakejahl, 2025-07-23:
You need to install gazebo harmonic or later

GibbekAdsk, 2025-07-23:
@dakejahl thx for the quick response, but how do I know which version I've already installed?
As mentioned above, a version of Gazebo is installed:
gz sim --versions => 8.9.0
gz gui --versions => 8.4.0
but I don't know if these version numbers belong to "Harmonic". I can't find anything about how to know if I've "Harmonic" installed or not.

If the version installed on my PC is not Harmonic or later, how can I uninstall it?

GibbekAdsk, 2025-07-23:
@dakejahl I just checked one more thing. when I start `gz sim` in my Ubuntu 22.04.5 LTS, then the dialog shows `Harmonic v 8.9.0` in the upper right corner => it seems `Harmonic` is installed.

[image]

dakejahl, 2025-07-23:
Hmm it must be a WSL problem then. I would suggest posting in the Discord to see if anyone uses WSL that can help. Otherwise you might want to work with an LLM and tinker around with the build

dakejahl, 2025-07-24:
Great!

closed 2025-07-24 by dakejahl