PX4/PX4-Autopilot #25273: [Bug] Offboard control setpoints affect vehicle when offboard control is not enabled

fennel-alpineeagle, 2025-07-23:
### Describe the bug

When the offboard control activation command (`VEHICLE_CMD_DO_SET_MODE` with parameters 1 and 6) is sent to a vehicle with non-matching system ID (as it happens, when the offboard control demo is used in conjunction with multiple simulation instances), offboard control is not enabled. This is a feature and right now I do not consider it as a bug. However, when setpoints are sent in this scenario via the `vehicle_command` and `trajectory_setpoint` at the same time, the setpoints do still affect the vehicle behavior, even though offboard control is not activated. Personally, I suspect that something with the setpoint multiplexing for offboard control is wrong.

### To Reproduce

Environment: PX4 v1.15 Gazebo SITL @ Ubuntu 22 with ROS Humble, but may work on other configurations.

1. Launch `MicroXRCEAgent udp4 -p 8888`
2. Launch `PX4_SYS_AUTOSTART=4004 PX4_SIM_MODEL=standard_vtol /root/PX4/build/px4_sitl_default/bin/px4 -i 3`
3. Take-off
4. Launch `ros2 run px4_ros_com offboard_control --ros-args --remap /fmu/in/offboard_control_mode:=/px4_3/fmu/in/offboard_control --remap /fmu/in/vehicle_command:=/px4_3/fmu/in/vehicle_command --remap /fmu/in/trajectory_setpoint:=/px4_3/fmu/in/trajectory_setpoint`
5. Observe how the drone starts to jitter in Gazebo, even though Offboard mode is not active according to QGC

Important here: PX4 is started with `-i 3`.

### Expected behavior

The vehicle should fully ignore the offboard setpoints, when offboard control was not properly activated.

### Flight Log

Not necessary, as reproducible in sim.

### Software Version

PX4 v1.15

### Flight controller

n/a

### Vehicle type

None

fennel-alpineeagle, 2025-07-31:
Update: recent experiments also revealed that the offboard control setpoints are affecting the system when offboard control is not engaged, independent of the system_id. Hence, this is not an edge case anymore. 

**This means, offboard control setpoints will interfere with the vehicle under internal control!**

varadVaidya, 2026-03-16:
@Pedro-Roque The gating should allow for RC to take over control right? Sufficient RC Stick deflection will cause mode to change to say Position Mode, stopping any further setpoints to be sent inside PX4. Or do you mean to say that incorrect usage of the soft gating will cause issues?