PX4/PX4-Autopilot #25279: mavlink iridium mode, is it used?

dakejahl, 2025-07-24:
### Describe the bug

I am working on fixing https://github.com/PX4/PX4-Autopilot/issues/25142 and doing a general overhaul of the mavlink module in order to implement a Token Bucket rate limiting system. The goal is for all mavlink microservices to use the token system to ensure we stay within bandwidth constraints. A common problem is parameter download failure on lossy or congested links. 

I see that there is special logic for the Iridium mode everywhere and it's not straightforward to break out. I first wanted to ask if anyone is still using Iridium modems and this mode? I see that @oravla5 did some work on it last year so I assume it's still useful https://github.com/PX4/PX4-Autopilot/pull/23048

@oravla5 @bkueng can you guys comment? The simplest thing would be to remove the mode altogether. If we can't do that perhaps breaking out the Iridium mode into it's own special main loop and functions, such that it doesn't clutter the more typical use case.

### Vehicle type

None

hamishwillee, 2025-07-24:
I assume you know that this is actually "very high latency mode". I don't know who (if anyone) is using it, but if you remove it then you effectively remove support for things like balloons, boats that really only need the occasional telemetry ping. Perhaps we're at the point where starlink makes this unnecessary, but maybe not.

dirksavage88, 2025-07-24:
@dakejahl it's still used, but it is somewhat broken and most have forked PX4 to fix it. I don't know of anyone using an up to date vanilla PX4 that uses it specifically.

dakejahl, 2025-07-24:
I see that this feature is documented
https://docs.px4.io/main/en/advanced_features/satcom_roadblock.html
And there is a dedicated iridium modem driver
https://github.com/PX4/PX4-Autopilot/tree/main/src/drivers/telemetry

I will keep the functionality, I just need to find a clean way to untangle it.

closed 2025-07-24 by dakejahl

ryanjAA, 2025-07-24:
Thanks @dirksavage88 

Ya, but you're right, forked and trying to get away from Iridium stuff anyway given how slow it is... Onboard companion is easier just doing the routing....