PX4/PX4-Autopilot #25345: [Bug] Flash Based Params - dataman and mavlink errors without SD Card

AlexKlimaj, 2025-07-31:
### Describe the bug

When using a flight controller that relies on flash for param storage, when the sd card is missing, will spam dataman and mavlink errors.

```
WARN  [mavlink] Operation timeout
ERROR [dataman_client] timeout after 5000 ms!
WARN  [mavlink] Operation timeout
ERROR [dataman_client] timeout after 5000 ms!
WARN  [mavlink] Operation timeout
ERROR [dataman_client] timeout after 5000 ms!
WARN  [mavlink] Operation timeout
ERROR [dataman_client] timeout after 5000 ms!
WARN  [mavlink] Operation timeout
WARN  [load_mon] navigator low on stack! (284 bytes left)
```

### Vehicle type

None

dagar, 2025-08-12:
Default dataman to RAM? Although the spamming isn't acceptable either way.

dagar, 2025-08-12:
Would it be better to have dataman fallback to RAM in this situation or fail all mission operations?

henrykotze, 2025-08-15:
In our use-case it would be to fail all missions operations: Using the ESP32 as a payload controller board.

closed 2026-08-15 by github-actions[bot]