PX4/PX4-Autopilot #25367: [Bug] Error: unknown target 'gz_x500'

lui-spa, 2025-08-05:
### Describe the bug

My goal is to run a basic simulation of a drone with px4/gazebo. My OS (Ubuntu 22.04.5 LTS (Jammy Jellyfish)) was just installed today for this project, so I had no previous installation/software. I followed the guides step by step: 
1) https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html#simulation-and-nuttx-pixhawk-targets
2) https://docs.px4.io/main/en/sim_gazebo_gz/

Unfortunately, I end up with error unknown target 'gz_x500' when trying to run make px4_sitl gz_x500 even if I already tried the proposed solution to this error from the guide: "If make px4_sitl gz_x500 gives the error ninja: error: unknown target 'gz_x500' then run make distclean to start from a clean slate, and try running make px4_sitl gz_x500 again."
and from other similar bug reports: https://github.com/PX4/PX4-Autopilot/issues/22728

FULL OUTPUT:

[full_output.txt](https://github.com/user-attachments/files/21603118/full_output.txt)

### To Reproduce

Official steps:
1. git clone https://github.com/PX4/PX4-Autopilot.git --recursive
2. bash ./PX4-Autopilot/Tools/setup/ubuntu.sh
3. restart PC
4. make distclean
5. make px4_sitl gz_x500

I repeated more times the steps 3/4/5 but nothing changed.
I also tried the solution from https://github.com/PX4/PX4-Autopilot/issues/22728:
export CC=gcc
export CXX=g++
Repeated again steps 3/4/5 but no change.
Also tried git submodule update --init --recursive, again no change.

### Expected behavior

Simulation tool opening.

### Screenshot / Media

Nothing to show, full output pasted above.

### Flight Log

N/A

### Software Version

I'm using the [last version of px4](https://github.com/PX4/PX4-Autopilot/tree/main)
PX4 version: v1.15.4
Ubuntu 22.04.5 LTS (Jammy Jellyfish)

### Vehicle type

None

farhangnaderi, 2025-08-05:
This happens to me as well. I do sumbodule update each time I do make distclean to solve it.

leelightning, 2025-08-11:
This happens to me as well. 
I use Ubuntu 24.04 and the latest version of PX4 source code. Gazebo 8 sim.

YukiDmZhao, 2025-08-11:
[image]

-- Build files have been written to: /home/asus/PX4-Autopilot/build/px4_sitl_default
ninja: error: unknown target 'gz_x500'
make: *** [Makefile:227: px4_sitl] Error 1
asus@LAPTOP-0C6R7USM:~/PX4-Autopilot$

closed 2025-08-15 by farhangnaderi

mrpollo, 2025-09-15:
This error typically happens when the build system cannot find the Gazebo dependencies.

hamishwillee, 2025-09-17:
Yes. The trick is to make sure you have included the development environment including gazebo: https://docs.px4.io/main/en/dev_setup/dev_env.html