PX4/PX4-Autopilot #25415: [Bug] Position control bounces all over, leading to crash

taile431909, 2025-08-13:
### Describe the bug

The drone was flying normally, then enters a chaotic oscillation phase, then crashes down. PID was tuned and working normally before.

### To Reproduce

1. Take off and flying normally.
2. Nothing was done, it just shakes violently.

### Expected behavior

Nothing, just normal flying.

### Screenshot / Media

### Flight Log

https://review.px4.io/plot_app?log=0a3c1eac-bae4-4214-a54c-791d303c05fe

### Software Version

1.15.4

### Flight controller

Holybro V6X-RT

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

Motors are controlled via PWM I/O. Telem1 is for MAVLINK, Telem2 is for SIYI A8 Mini control. GPS 1 has a Holybro M9N GPS.

### Additional context

It was flying normally before.

taile431909, 2025-08-13:
(https://drive.google.com/file/d/1-Bzh-P0WU0RkIEuB0jxW0WJ07wprNKar/view?usp=sharing)
Video of the crash.

PetervdPerk-NXP, 2025-08-15:
```
WARN  [health_and_arming_checks] Preflight: GPS Horizontal Pos Error too high
WARN  [health_and_arming_checks] Preflight Fail: height estimate not stable
WARN  [health_and_arming_checks] Preflight: GPS Horizontal Pos Error too high
WARN  [health_and_arming_checks] Preflight Fail: height estimate not stable
WARN  [health_and_arming_checks] Preflight: GPS Horizontal Pos Error too high
WARN  [health_and_arming_checks] Preflight: GPS Horizontal Pos Error too high
WARN  [health_and_arming_checks] Preflight Fail: Compasses 125° inconsistent
WARN  [health_and_arming_checks] Preflight: GPS Horizontal Pos Drift too high
WARN  [health_and_arming_checks] Preflight Fail: Compasses 136° inconsistent
WARN  [health_and_arming_checks] Preflight Fail: Compasses 139° inconsistent
WARN  [health_and_arming_checks] Preflight Fail: height estimate not stable
WARN  [health_and_arming_checks] Preflight Fail: Compasses 127° inconsistent
WARN  [health_and_arming_checks] Preflight Fail: Compasses 130° inconsistent
WARN  [health_and_arming_checks] Preflight Fail: Compasses 129° inconsistent
```

Could you check if the orientation of FMU and GPS/Compass matches? Preflight seems to indicate that it's not.
Also it seems that GPS fix might be unstable.