PX4/PX4-Autopilot #25488: [Bug] PWM Parachutes (and other safty divices) doesn't function correctly

F00dP00l, 2025-08-26:
### Describe the bug

According to the document, triggering failsafe(in PX4, the **TERMINATE** status) will set all the PWM channels to PWM_MAIN_FAIL*, therefore opening the parachutes (or any other safty devices). However, with the code below
`MixingOutput::limitAndUpdateOutputs(float outputs[MAX_ACTUATORS], bool has_updates)
{
	if (_armed.lockdown || _armed.kill) {
		// overwrite outputs in case of lockdown with disarmed values
		for (size_t i = 0; i < _max_num_outputs; i++) {
			_current_output_value[i] = _disarmed_value[i];
		}

	} else if (_armed.termination) {
		// Overwrite outputs with _failsafe_value when terminated
		for (size_t i = 0; i < _max_num_outputs; i++) {
			_current_output_value[i] = actualFailsafeValue(i);
		}

	} else {
		// the output limit call takes care of out of band errors, NaN and constrains
		output_limit_calc(_throttle_armed || _actuator_test.inTestMode(), _max_num_outputs, outputs);
	}
        ......
}`
in "Ardupilot/src/lib/mixer_module/mixer_module.cpp", PX4 will detect **lockdown** or **kill** status before **termination** status (in previrous versions, this var is named **force_failsafe**), which cause PWM outputs be set to PWM_MAIN_DIS* instead of the expected PWM_MAIN_FAIL*. It causes a major safty issue during protected flights. Simply change the sequence of judgement will fix the error and it works perfectly well on my own drone.

### Vehicle type

None

dakejahl, 2025-08-27:
Can you provide a log or steps to reproduce the issue for verification? Also if it's indeed broken then a PR to fix the logic would be appreciated!

F00dP00l, 2025-08-28:
[log_6_2025-8-26-10-29-18_actuator_outputs_0.csv](https://github.com/user-attachments/files/22019082/log_6_2025-8-26-10-29-18_actuator_outputs_0.csv)
[log_6_2025-8-26-10-29-18_actuator_outputs_1.csv](https://github.com/user-attachments/files/22019080/log_6_2025-8-26-10-29-18_actuator_outputs_1.csv)
[log_6_2025-8-26-10-29-18_vehicle_status_0.csv](https://github.com/user-attachments/files/22019081/log_6_2025-8-26-10-29-18_vehicle_status_0.csv)

It can be seen that the *output[4]* remains to be 1000us during the flight, even after terminated, which cause the parachute didn't work correctly.

dakejahl, 2025-08-28:
link to the ulog please

F00dP00l, 2025-08-29:
[logs](https://github.com/F00dP00l/log_temp)