PX4/PX4-Autopilot #25491: FLASH savings feature tracker

dakejahl, 2025-08-26:
### Describe the bug

I am creating this issue so that we can keep track of a list of options for reducing FLASH usage throughout the project. Please feel free to edit this description to update the list.

- Split builds by airframe type
- Compile time airframe selection (eg /etc/init.d/airframes/4050_generic_250)
- Compile time rcS and compile time module start hooks
- Selectively compile mavlink messages/features (context here https://github.com/PX4/PX4-Autopilot/pull/22930)
- Split GNSS drivers by module type (kill gps.cpp monolith)

### Vehicle type

None

dirksavage88, 2025-09-02:
@dakejahl is there a way to profile where the majority of flash usage is coming from? For example on the v6x. I am wondering if it is worth also going down to the common x (driver level) and disabling some of these lesser used drivers. Also taking a granular look at each driver and seeing if any flash savings could be had. Similar to what was done for the common code across the IMU drivers, but applied to peripheral sensors.

alexcekay, 2025-09-02:
@dirksavage88: I created this just a few days ago. https://docs.google.com/spreadsheets/d/1Gh-S_VSWtpxTG43m1hhNQfr3SG523hlGoGV1z8SUOt0/edit?usp=sharing

Can share the tooling if you guys are interested.

dakejahl, 2025-12-12:
Bump.

Splitting builds by airframe type - we talked about it but who wants to do it?

farhangnaderi, 2025-12-17:
@dakejahl Is there any PR relevant to this going on? Either PX4/QGC?

closed 2026-09-17 by github-actions[bot]