PX4/PX4-Autopilot #25531: [Bug] PositionSetpointTriplet.msg May Not Comply with uORB IDL Convention

shouhengyi, 2025-09-08:
### Describe the bug

Please see the last three fields in https://github.com/PX4/PX4-Autopilot/blob/main/msg/PositionSetpointTriplet.msg#L6

```
PositionSetpoint previous
PositionSetpoint current
PositionSetpoint next
```

To my knowledge, those are ROS .msg IDL convention, not the uORB field definition format. This might not be an issue. It could just be my lack of knowledge on the uORB IDL grammar. Feel free to ignore.

### Vehicle type

None

dakejahl, 2025-09-08:
This is the syntax of nested uORB message definitions. See EscStatus.EscReport as another example. What is the problem?

shouhengyi, 2025-09-08:
Found it: https://px4.io/px4-uorb-explained-part-4-ulog-flight-logging-system/

closed 2025-09-08 by shouhengyi