PX4/PX4-Autopilot #25575: PX4-Gazebo Model Respawn

eOvic, 2025-09-16:
### Describe problem solved by the proposed feature

I want to use RL with PX4 and Gazebo but sometimes the drone crashes and i need to reset the world, the drone model but when the drone is deleted from the gazebo the connection from PX4 breaks. Any solutions ?

### Describe your preferred solution

.

### Describe possible alternatives

.

eOvic, 2025-09-17:
Hello @Siri2K well you see when doing RL sometimes the drone crashes and flips so like in the gym env I need to reset the training, the simulation needs to be reset. This is ez in normal case if I use a model not connected to PX4, but once the PX4 sitl start removing a drone and spawning another one causes some conflicts

eOvic, 2025-09-17:
@Siri2K  have a look: https://github.com/PX4/PX4-Autopilot/issues/12590

farhangnaderi, 2025-09-17:
@eOvic saving your RL model in each training episode and continue training running PX4 again from the beginning wont work for you?