PX4/PX4-Autopilot #25585: [Bug] GPS shows as "Disabled" instead of "Enabled" in QGC

ZoneDK-sudo, 2025-09-18:
### Describe the bug

When powering on the vehicle, QGroundControl shows `GPS: Disabled` in the Sensor Status panel.  
Normally, I expect QGC to display GPS status (e.g. "OK", "No Fix", "3D Fix"), but instead it is always showing as Disabled.

### To Reproduce

1. Power on the drone with PX4 firmware.
2. Connect QGroundControl.
3. Open the Vehicle Setup > Summary / Sensors tab.
4. Observe that GPS sensor status is listed as "Disabled".

### Expected behavior

QGC should show GPS state normally (e.g. "OK", "No Fix", "3D Fix").  
If GPS is explicitly disabled by configuration (e.g. parameter setting), there should be a clear message rather than just "Disabled".

### Screenshot / Media

[image]

### Flight Log

Log
https://review.px4.io/plot_app?log=51e97040-7704-4929-8e23-ea5a924d96c4
older logs
https://drive.google.com/drive/folders/1H3_N_DTPb_jf7AvfcH6K5BPruZcvuVst?usp=sharing
Parameters
https://drive.google.com/drive/folders/14XVjBgypfK8Hm1U-w_MyrOTfSMmMgyKB?usp=sharing

### Software Version

PX4 1.16.0

### Flight controller

Pixhawk mini 6c

### Vehicle type

Hybrid VTOL

### Additional context

Here 3 GPS RTK

dakejahl, 2025-09-18:
The message above is telling you what's wrong

[image]

ZoneDK-sudo, 2025-09-18:
> The message above is telling you what's wrong
> 
> <img alt="Image" width="334" height="92" src="https://private-user-images.githubusercontent.com/37091262/491291093-155138e6-1c2d-43a1-b874-d8e698ae8f62.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTgyMzAyNjgsIm5iZiI6MTc1ODIyOTk2OCwicGF0aCI6Ii8zNzA5MTI2Mi80OTEyOTEwOTMtMTU1MTM4ZTYtMWMyZC00M2ExLWI4NzQtZDhlNjk4YWU4ZjYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTE4VDIxMTI0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTExMDFhZGQxYjRkMzZkZTU5OWU2OWJkZjFlZjhkZmFiNDMwMTRlMzViYTYyYjY1MjJlZThlNWUxMzZlOGNlMDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.p7A5eY9fAvk1lXhmn3cwk380tO_fFKdPoD0tOB8VmcY">

sensor status still says disabled below

dakejahl, 2025-09-18:
It looks like you have GPS interference. The GPS is reporting NED speeds all over the place. This is causing the GPS preflight checks to fail. See [EKF2_GPS_CHECK](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#EKF2_GPS_CHECK)

ZoneDK-sudo, 2025-09-18:
[image]

ZoneDK-sudo, 2025-09-19:
[image]

is there a log that I can upload and help to troubleshoot/resolve these issues?

closed 2025-09-19 by dakejahl